Lauszus/Sanguino #6: burning bootloader issue

charizord, 2015-05-10:
I get this error:
avrdude: ser_open(): can't open device "\\.\com1": The system cannot find the file specified.

when trying to burn the bootloader on my atmega644p through arduino nano as isp

Lauszus, 2015-05-10:
Are you sure that you have selected the right com port?

charizord, 2015-05-10:
Yup, Com4. when I upload programs there is no issue (to the nano not the 644P).

charizord, 2015-05-10:
I got it working by selecting arduino as isp second option. but now when the bootloader is running
I get this eror:
avrdude: Expected signature for ATmega644P is 1E 96 0A
         Double check chip, or use -F to override this check.
Wrong microcontroller found.  Did you select the right board from the Tools > Board menu?

and it stops

Lauszus, 2015-05-10:
Yes you need "Arduino as ISP" and the Sanguino as the board.

Can you post your full error output?

charizord, 2015-05-10:
that is the full error output

Lauszus, 2015-05-10:
No it's not. It should print the read signature as well. Something like this:
```
avrdude: Device signature = 0xXXXXXX
```

charizord, 2015-05-10:
Ok, I fixed my arduino settings. Here is the full error:

Arduino: 1.6.3 (Windows 7), Board: "Sanguino, ATmega644P or ATmega644PA (16 MHz)"

C:\Users\Ehud\AppData\Roaming\Arduino15\packages\arduino\tools\avrdude\6.0.1-arduino2/bin/avrdude -CC:\Users\Ehud\AppData\Roaming\Arduino15\packages\arduino\tools\avrdude\6.0.1-arduino2/etc/avrdude.conf -v -patmega644p -cstk500v1 -PCOM4 -b19200 -e -Ulock:w:0x3F:m -Uefuse:w:0xFD:m -Uhfuse:w:0xDE:m -Ulfuse:w:0xFF:m 

avrdude: Version 6.0.1, compiled on Mar 19 2015 at 14:43:01

         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/

         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Users\Ehud\AppData\Roaming\Arduino15\packages\arduino\tools\avrdude\6.0.1-arduino2/etc/avrdude.conf"

         Using Port                    : COM4

         Using Programmer              : stk500v1

         Overriding Baud Rate          : 19200

         AVR Part                      : ATmega644P

         Chip Erase delay              : 55000 us

         PAGEL                         : PD7

         BS2                           : PA0

         RESET disposition             : dedicated

         RETRY pulse                   : SCK

         serial program mode           : yes

         parallel program mode         : yes

         Timeout                       : 200

         StabDelay                     : 100

         CmdexeDelay                   : 25

         SyncLoops                     : 32

         ByteDelay                     : 0

         PollIndex                     : 3

         PollValue                     : 0x53

         Memory Detail                 :

                                  Block Poll               Page                       Polled

           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack

           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------

           eeprom        65    10   128    0 no       2048    8      0  9000  9000 0xff 0xff

           flash         33     6   256    0 yes     65536  256    256  4500  4500 0xff 0xff

           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00

           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00

           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00

           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00

           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : STK500

         Description     : Atmel STK500 Version 1.x firmware

         Hardware Version: 2

         Firmware Version: 1.18

         Topcard         : Unknown

         Vtarget         : 0.0 V

         Varef           : 0.0 V

         Oscillator      : Off

         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.05s

avrdude: Device signature = 0x1e9609

avrdude: Expected signature for ATmega644P is 1E 96 0A

         Double check chip, or use -F to override this check.

avrdude done.  Thank you.

Wrong microcontroller found.  Did you select the right board from the Tools > Board menu?

charizord, 2015-05-10:
does my atmega644 being PU-20 and not P make a difference?

Lauszus, 2015-05-10:
Based on the signature. You will need to select ATmega644.

charizord, 2015-05-10:
It worked thank you, I will now move the atmega to my circuit, connect to my ftdi chip and will report back if everything works.

Lauszus, 2015-05-10:
You are welcome.

Are you sure that it was actually done twice and it's not just the verification step you see the second time?

closed 2015-05-10 by Lauszus

charizord, 2015-05-10:
They look quite similar, but what do I know :)

C:\Users\Ehud\AppData\Roaming\Arduino15\packages\arduino\tools\avrdude\6.0.1-arduino2/bin/avrdude -CC:\Users\Ehud\AppData\Roaming\Arduino15\packages\arduino\tools\avrdude\6.0.1-arduino2/etc/avrdude.conf -v -patmega644 -cstk500v1 -PCOM4 -b19200 -e -Ulock:w:0x3F:m -Uefuse:w:0xFD:m -Uhfuse:w:0xDE:m -Ulfuse:w:0xFF:m 

avrdude: Version 6.0.1, compiled on Mar 19 2015 at 14:43:01
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Users\Ehud\AppData\Roaming\Arduino15\packages\arduino\tools\avrdude\6.0.1-arduino2/etc/avrdude.conf"

         Using Port                    : COM4
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega644
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       2048    8      0  9000  9000 0xff 0xff
           flash         33     6   256    0 yes     65536  256    256  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.06s

avrdude: Device signature = 0x1e9609
avrdude: erasing chip
avrdude: reading input file "0x3F"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3F:
avrdude: load data lock data from input file 0x3F:
avrdude: input file 0x3F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.04s
[... 116 lines cut ...]
avrdude: load data flash data from input file C:\Program Files (x86)\Arduino\hardware\Sanguino-master\avr/bootloaders/optiboot/optiboot_atmega644.hex:
avrdude: input file C:\Program Files (x86)\Arduino\hardware\Sanguino-master\avr/bootloaders/optiboot/optiboot_atmega644.hex contains 65536 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% -0.00s

avrdude: verifying ...
avrdude: 65536 bytes of flash verified
avrdude: reading input file "0x0F"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.07s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0F:
avrdude: load data lock data from input file 0x0F:
avrdude: input file 0x0F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.

Lauszus, 2015-05-10:
If you look closely, you will see that it first sets the fuses, then burns the bootloader.