Lauszus/Sanguino #8: Unable to Burn Bootloader on a Sanguino ATmega644P

mwalz, 2016-01-11:
Hello,

I'm unable to Burn Bootloader on a Sanguino ATmega644P.

Settings:
Board: Sanguino
Processor: ATmega644P or ATmega644PA (16 MHz)
Programmer: Arduino as ISP

Serial Monitor does appear to work, it shows 'start' from the Arduino when it connects. This if for an original Huxley RepRap from late 2011 / early 2012. 

I keep getting the following output:

avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x39
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x3c
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xac
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0xff
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xff
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xff
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xff
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xff
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xff

I've tried multple Programmer settings with no success, do you have any suggestions?

Thanks!

Lauszus, 2016-01-11:
Make sure that you have wired the ISP programmer correctly. Please see the following page: https://www.arduino.cc/en/Tutorial/ArduinoToBreadboard.
Also have you disabled auto-reset on the Arduino you are trying to use as ISP?

Lauszus, 2016-01-18:
@mwalz any updates?

mwalz, 2016-01-26:
Apologies for the delayed response. I did have the jumper in place. I believe that was causing issues. In the process of removing it, the board was short circuited. I have another board on the way and hope to test is in the next few days.

Thanks for your initial answer and follow up. I will update by the end of this week.

Lauszus, 2016-01-26:
@mwalz okay no problem. Please reopen if needed :)

closed 2016-01-26 by Lauszus

nkuck, 2016-06-05:
I am experiencing the same problem (avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x66) after unsuccessfully trying to upload Marlin FW to my Melzi board. All worked fine before. I can upload a sketch to a different Arduino board and all is fine. But when I try to upload Marlin to my printer, I get the message. Any suggestions? Thank for any direction. Could the boot loader have been corrupted? I can still print.

nkuck, 2016-06-05:
Never mind. My error. Had the jumper pin connected on the board. All is good.

nkuck, 2016-06-28:
Please do mind : ).
I thought that it was working but nope. Getting the common not in sync error. Jumper is in place and the board works fine as-is; just can't upload newer firmware. Ideas? Thanks any/everyone.

Lauszus, 2016-07-02:
Have you tried to re-burn the bootloader?

nkuck, 2016-07-02:
Not tried the bootloader as I read about it being complex.

Live simply!
You can only take with you what you have given.

Lauszus, 2016-07-02:
This add-on will not work with the standard bootloader, as this uses the improved optiboot bootloader.

nkuck, 2016-07-02:
Are you suggesting that I attempt the optiboot bootloader?

Live simply!
You can only take with you what you have given.

Lauszus, 2016-07-02:
Yes. You need to install the add-on. Select the Sanguino and then burn the bootloaders. You should be able to find guides on how to burn the bootloader using a second Arduino online.

nkuck, 2016-07-03:
Can I use the Arduino IDE with the computer plugged into the Sanguino board to simply upload the optiboot file to it?

Lauszus, 2016-07-03:
No you will to use a second Arduino as a programmer. Please see: https://www.arduino.cc/en/Tutorial/ArduinoISP.

nkuck, 2016-07-03:
But won't that require me unsoldering the chip from the Sanduino board? Am I confused?

Live simply!

Lauszus, 2016-07-03:
No it doesn't. You simply connect to the header labeled ICSP on the Sanguino board.

nkuck, 2016-07-03:
Oh, got it. Will advise when successful. So, burn the bootloader then able to load the new firmware? Thanks for your support. 

Live simply!
You can only take with you what you have given.

Lauszus, 2016-07-03:
Okay good. Yes exactly :)

nkuck, 2016-07-04:
Oh, oh, I may need further handholding, like step-by-step directions. What connects to the ICSP pins on the Sanguino board? Sorry😱.

Live simply!
You can only take with you what you have given.

Lauszus, 2016-07-04:
Sorry I don't have time to do that. Please see: https://www.arduino.cc/en/Tutorial/ArduinoISP and https://learn.sparkfun.com/tutorials/installing-an-arduino-bootloader.

comarius, 2017-04-09:
DON'T BUY  Sanguino  boards !!! Support sucks.

nkuck, 2017-04-10:
Agree!

Noel Kuck

Live simply!

Lauszus, 2017-04-17:
@comarius @nkuck please note that I am not affiliated with the people behind Sanguino. I simply published this repository, as I found that the official board add-on was outdated.

Lauszus, 2018-03-15:
@comarius and @nkuck please note that it is now also possible to use this add-on with PlatformIO. This will allow you to upload to the board without having to upload the bootloader. Please see the instructions in the readme: https://github.com/Lauszus/Sanguino#platformio.