Lauszus/Sanguino #10: Error trying to upload Marlin

niksoley, 2016-10-26:
I have a gen6.d board, and I am trying to upload a newer version of Marlin to it, but I am receiving the following error:
"avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x0e
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x3a
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x48
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x54
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x4a
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x40
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xc2
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x2c
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x68
Problem uploading to board."

The board is working fine for printing, the update aims to enable auto bed leveling. 
The Sanguino was installed by Boards Manager.
I have no compilation problem, and Serial Monitor Echo is able to pickup data from the board as follow:
"start
echo:Free Memory:799 PlannerBufferBytes:1168
echo:Using Default settings:
echoconfused smileyteps per unit:
echo: M92 X40.00 Y40.00 Z2078.72 E514.00
echo:Maximum feedrates (mm/s):
echo: M203 X250.00 Y250.00 Z3.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo: M201 X4500 Y4500 Z40 E5000
echo:Acceleration: S=acceleration, T=retract acceleration
echo: M204 S1500.00 T1500.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum xY jerk (mm/s), Z=maximum Z jerk (mm/s)
echo: M205 S0.00 T0.00 B20000 X10.00 Z0.20
echotongue sticking out smileyID settings:
echo: M301 P22.20 I1.25 D99.00
echoconfused smileyD init fail
workDir open failed"

About my specs:
MacOS Sierra 10.12.1;
Gen6.d board;
IDE Arduino 1.6.12 with Board: "Sanguino", Processor: ATmega1284 or ATmega1284P (16 MHz), Port: "dev/cu.usbserial-A101OV2J";
Sanguino 1.0.1;
USB Driver: FT232R USB UART;

Tks in adv.

Lauszus, 2016-10-27:
You need to re-burn the bootloader. Please see the following issue: https://github.com/Lauszus/Sanguino/issues/2.

Lauszus, 2016-11-02:
Did you try to re-burn the bootloader yet?

niksoley, 2016-11-04:
Just bought an arduino, and will try it this weekend. I will update the issue as soon as possible.

niksoley, 2016-11-09:
Hey Lauszus,

Just did it without burn the bootloader, it was a configuration in the board.txt, the resolution was to change atmega644p.upload.speed=115200 to 38400

Lauszus, 2016-11-10:
Okay. If you want faster upload speeds, then I would recommend burning the bootloader from this repository. Anyway I will close this issue, as I believe your issue has been solved :)

closed 2016-11-10 by Lauszus

lukewakeford, 2020-06-20:
I've got the same issue as above, Recently purchased an old used prusa mega, with gen6 electronics - appears to have ATMEGA644P chip, trying upload new firmware and or burn boot loader hits me with the following:

```avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x03

... etc all the way to 10 of 10
```

250000 baud rate appears to work for pronterface and monitoring elsewhere, not sure why the boot loader burn forces something else.

Am I missing something obvious?

Lauszus, 2020-06-22:
@lukewakeford I would say that it's 99.9 % likely that there is already some kind of bootloader on your board. The easiest for you would probably be to use another Arduino to flash the bootloader bundled with this plugin.

You can find more information on how to use an Arduino as an ISP here: https://www.arduino.cc/en/tutorial/arduinoISP.

lukewakeford, 2020-06-22:
Hey thanks for the reply,

I'd actually been looking in to ArduinoISP thismorning.. tho the only other boards i have are ATtiny 85 - which don't appear to be big enough?

Might be simpler to just buy a ATMega and relatively newer RAMPs combo than battling with this?

Guess there isn't an easy way to suss out which bootloader was used to program this gen-6 board originally?

Lauszus, 2020-06-22:
> I'd actually been looking in to ArduinoISP thismorning.. tho the only other boards i have are ATtiny 85 - which don't appear to be big enough?

Nope. You need something like an Arduino Uno.

> Might be simpler to just buy a ATMega and relatively newer RAMPs combo than battling with this?

That's up to you. Though an Arduino Uno is always good to have around :)

> Guess there isn't an easy way to suss out which bootloader was used to program this gen-6 board originally?

You can change try all the different standard baudrate by changing the following line in [boards.txt](https://github.com/Lauszus/Sanguino/blob/master/boards.txt):

```bash
atmega644p.upload.speed=115200
```

lukewakeford, 2020-06-22:
Arduino Uno Ordered 👌 

Are you able to point me at a list of baud rates to try in the meantime?

Also, once i have the uno and ArduinoISP flashed, should it be as simple as using this lib to burn the new bootloader and then bobs my uncle?

Lauszus, 2020-06-22:
> Arduino Uno Ordered 👌 

Great. 

> Are you able to point me at a list of baud rates to try in the meantime?

I would try: 9600, 19200, 38400, 57600 and 115200.

> Also, once i have the uno and ArduinoISP flashed, should it be as simple as using this lib to burn the new bootloader and then bobs my uncle?

It's all explained in the link I sent earlier: https://www.arduino.cc/en/tutorial/arduinoISP :)

lukewakeford, 2020-06-22:
Cheers, will let you know how I get on :)

lukewakeford, 2020-06-23:
> I would try: 9600, 19200, 38400, 57600 and 115200.

None of these worked btw, wondering if there is no bootloader!? would that be odd?
Any others i can try?

Uno should be here soon 🤞

Lauszus, 2020-06-24:
You could also try 250000.

lukewakeford, 2020-06-24:
> An error occurred while uploading the sketch
> avrdude: serial_baud_lookup(): Using non-standard baud rate: 250000avrdude: ser_setspeed(): tcsetattr() failed
> avrdude: ser_open(): can't set attributes for device "/dev/cu.usbserial-AD02D60T": Invalid argument
> 
> avrdude done.  Thank you.

Guess thats a no then - FYI I tried the reset button trick for all these baud rates as well as without.

lukewakeford, 2020-06-24:
Uno Arrived.

```avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/Users/luke/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.wchusbserial14130
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega644P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       2048    8      0  9000  9000 0xff 0xff
           flash         33     6   256    0 yes     65536  256    256  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.08s

avrdude: Device signature = 0x1e960a (probably m644p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/Users/luke/Library/Arduino15/packages/Sanguino/hardware/avr/1.0.3/bootloaders/optiboot/optiboot_atmega644p.hex"
avrdude: writing flash (65536 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 65536 bytes of flash written
avrdude: verifying flash memory against /Users/luke/Library/Arduino15/packages/Sanguino/hardware/avr/1.0.3/bootloaders/optiboot/optiboot_atmega644p.hex:
avrdude: load data flash data from input file /Users/luke/Library/Arduino15/packages/Sanguino/hardware/avr/1.0.3/bootloaders/optiboot/optiboot_atmega644p.hex:
avrdude: input file /Users/luke/Library/Arduino15/packages/Sanguino/hardware/avr/1.0.3/bootloaders/optiboot/optiboot_atmega644p.hex contains 65536 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 65536 bytes of flash verified
avrdude: reading input file "0x0F"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.08s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0F:
avrdude: load data lock data from input file 0x0F:
avrdude: input file 0x0F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.03s

avrdude: verifying ...
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.```

Looks like the bootloader burnt successfully.. guess ill see if it accepts marlin now 😬

lukewakeford, 2020-06-24:
🔥 🔥 👍 ⚡ 🥳 🎉 

```
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/Users/luke/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbserial-AD02D60T
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega644P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       2048    8      0  9000  9000 0xff 0xff
           flash         33     6   256    0 yes     65536  256    256  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 5.0
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e960a (probably m644p)
avrdude: reading input file "/var/folders/hx/5rfqshhs1mz04d_p196sxm9m0000gn/T/arduino_build_158746/Marlin.ino.hex"
avrdude: writing flash (52408 bytes):

Writing | ################################################## | 100% 9.83s

avrdude: 52408 bytes of flash written
avrdude: verifying flash memory against /var/folders/hx/5rfqshhs1mz04d_p196sxm9m0000gn/T/arduino_build_158746/Marlin.ino.hex:
avrdude: load data flash data from input file /var/folders/hx/5rfqshhs1mz04d_p196sxm9m0000gn/T/arduino_build_158746/Marlin.ino.hex:
avrdude: input file /var/folders/hx/5rfqshhs1mz04d_p196sxm9m0000gn/T/arduino_build_158746/Marlin.ino.hex contains 52408 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 11.02s

avrdude: verifying ...
avrdude: 52408 bytes of flash verified

avrdude done.  Thank you.
```

Apologies for the triple post but that worked a right treat! many thanks.

My Printer now sounds a million times quieter!!!!!

Lauszus, 2020-06-24:
Wuhuu, great to hear! Happy printing :)