Lauszus/Sanguino #12: Using Sanguino from command-line / Travis CI

thinkyhead, 2017-09-25:
Hi! I'm trying to do continuous integration testing with some code that targets Sanguino, and I can't get it to build. Here's the code that is being executed under Travis CI for my project, and its output:

[image]

Can you offer some guidance how I can get an Arduino install to work properly with this hardware profile? It has become the de-facto choice for those who want to build in the IDE, and within the IDE I have no trouble using it. But it would be nice to be able to have Travis CI also include the Sanguino platform, as we still target that with our project (due mainly to the popularity of Melzi).

Thanks for any help you can provide!

Lauszus, 2017-09-25:
I would recommend using the following script instead: https://github.com/adafruit/travis-ci-arduino.

This is what I have been using for the Balanduino: https://github.com/TKJElectronics/Balanduino/blob/master/.travis.yml.

If you need a newer version of Arduino, then check out this fork: https://github.com/fede2cr/travis-ci-arduino.

Lauszus, 2017-10-17:
@thinkyhead I am now using Travis to compile the Blink example for all the boards: https://github.com/Lauszus/Sanguino/commit/da82c384908ba6aebf3d7d58a7c486ad42974557.

closed 2017-10-17 by Lauszus