Lauszus/Sanguino #13: Burning optiboot onto AtMega1284p

bummer6, 2017-11-02:
Hello! I'm a bit of a noob at this, but I've run out of options, so I decided to post my problem here. I have a Sanguinololu 1.3a and I'm trying to burn the bootloader to it so I can flash the firmware. The bootloader burns fine, but then flashing the firmware doesn't work, giving me errors like this:

avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xe3

So I assume it's a problem where optiboot doesn't support AtMega1284p, and I see there are instructions how to burn optiboot properly onto the board, but the problem is that I don't understand the instructions at all. Here's what the instructions say;

To burn the bootloader for the ATmega1284 running at 16MHz use the following command:

make atmega1284p_isp ISPPORT=/dev/tty.usbmodem26411

Now... It says you use the following command, but where? Where do I enter the command, and what do I do afterwards?

I'm on Windows 10 if that matters... I'm out of options and I don't know what to do at this point.

Patrik356b, 2017-11-03:
That is a command for "avr dude" iirc. Which is a CLI. (Command-Line Interface)
been 2 years since i visited this project so my memory is faint.

Lauszus, 2017-11-03:
@bummer6 there are a number of guides online for using avrdude to burn the bootloader. Please read some of them for more information: http://www.ladyada.net/learn/avr/avrdude.html, https://learn.adafruit.com/usbtinyisp/avrdude# and https://learn.sparkfun.com/tutorials/pocket-avr-programmer-hookup-guide/using-avrdude.

However this is all automated by the [Makefile](https://github.com/Lauszus/Sanguino/blob/master/bootloaders/optiboot/Makefile). Anyway since you are using Windows 10, you can install it pretty easy using bash: https://www.howtogeek.com/249966/how-to-install-and-use-the-linux-bash-shell-on-windows-10/. Then run the following command in a terminal:

```bash
sudo apt-get install build-essential avrdude
```

Note that I am not using Windows, so I have actually not tested the command above.

That should install everything you need. Now navigate to [bootloaders/optiboot](https://github.com/Lauszus/Sanguino/blob/master/bootloaders/optiboot) in a terminal and execute the command described in [bootloaders/optiboot/ATmega1284.md](https://github.com/Lauszus/Sanguino/blob/master/bootloaders/optiboot/ATmega1284.md). Remember to add the ```-F``` flag in [Makefile.isp](https://github.com/Lauszus/Sanguino/blob/master/bootloaders/optiboot/Makefile.isp) as well.

Lauszus, 2017-11-09:
@bummer6 the easiest way is just to burn the bootloader via the Arduino IDE by using a second Arduino as a programmer. Please see: https://www.arduino.cc/en/Tutorial/ArduinoISP. You do not need to use the command line at all!

Please note that those special instructions are only for the "ATmega1284" and not the "ATmega1284p". You only need to use the command line if you want to burn the bootloader onto a "ATmega1284".

81ninja, 2017-12-23:
I'm having the same problem with a Sanguinolulu. I have a Geeetech 3D printer that uses it, and I'm trying to update the Marlin firmware on it to the latest version.

I can upload firmware to it fine using Arduino 1.0.1 (via the IDE), using a hardware package for the 1284(p) provided by Geeetech (from an unknown source). Marlin however requires Arduino 1.6+, and the old package of course won't work with it. Neither will this one - I can compile the sketch, but get the stk500_getsync() errors during uploading.

I can understand makefiles etc., but I'm not that familiar with Arduino yet, and the whole thing is still ahead of me. Since I can upload using Arduino 1.0.1, and the old package also includes bootloader files etc. Do you think it could be made to work with Arduino 1.6?

Lauszus, 2018-01-01:
@81ninja good to hear you got it working :)

Lauszus, 2018-01-01:
@bummer6 I will close this issue now due to lack of response. Feel free to open it again.

closed 2018-01-01 by Lauszus