Lauszus/Sanguino #15: Problem loading firmware to Melzi board

nexgen91, 2018-09-12:
I am having the hardest time sending firmware to my melzi board using Uno and IDE 1.8.6. I keep getting the following. I have removed all file paths from the log, but everything else should be there, let me know if you need something else.
My Configuration.h file contains the following which I assume reference the board type:
#define MOTHERBOARD 63

========== Start configuration string ==========
    "motherboard": 63,

I am using the following settings:
Board: Sanguino
ATmega1284 or ATmega1284P (16MHz)
Arduino as ISP

> Arduino: 1.8.6 (Windows 10), Board: "Sanguino, ATmega1284 or ATmega1284P (16 MHz)"

Sketch uses 91438 bytes (70%) of program storage space. Maximum is 130048 bytes.
Global variables use 4065 bytes (24%) of dynamic memory, leaving 12319 bytes for local variables. Maximum is 16384 bytes.

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "D:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM7
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega1284P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         65    10   256    0 yes    131072  256    512  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 4.4
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

An error occurred while uploading the sketch
Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: Expected signature for ATmega1284P is 1E 97 05
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.

nexgen91, 2018-09-12:
I think I fixed it, I was not using "Upload using programmer" that seems to have solved the problem.

closed 2018-09-12 by nexgen91

Lauszus, 2018-09-12:
Good to hear you solved it yourself :)