Lauszus/Sanguino #16: Is this a warning

comarius, 2018-09-30:
AFter importing the json for new board.
OS: linux Ubuntu 16.04
arduino ide: 1.8.7
[image]

per1234, 2018-10-03:
You've completely screwed up on the installation. Did you even read the instructions in the readme?

Do this:
1. To get rid of your bad installation job, delete ~/APPS/arduino-1.8.7/hardware/Sanguino
2. (In the Arduino IDE) Tools > Board > Boards Manager
3. Wait for downloads to finish
4. Scroll down until you see "Sanguino by Kristian Sloth Lauszus". Click on it.
5. Click "Install".
6. Wait for installation to finish.
7. Click "Close".
8. Tools > Board > Sanguino
9. Select the appropriate **Tools > Processor** selection for your hardware.

Lauszus, 2018-10-18:
@comarius as @per1234 said it looks like you have messed up the installation. There is no need to add or modify any files. Please just add the URL and then install the board using the boards manager.

Btw it automatically uses the platform.txt bundled with Arduino due to this line:

https://github.com/Lauszus/Sanguino/blob/8c7336114223bb9e7e19ac61b1666b3e5a807806/avr/boards.txt#L18

So that is the reason why it's not needed in this repo.

closed 2018-10-18 by Lauszus