Lauszus/Sanguino #19: Probleme téléversement

kinou33, 2019-09-08:
Bonjour
j'ai une ender 3 avec une carte mère 1.1.5 qui fonctionne parfaitement " vive le silence" .
j'ai installé touch-mi dessus et au moment de téléversé marlin 1.1.9 avec touch-mi j’obtiens une erreur .
j'ai cherché partout ...essayer plusieurs méthodes.... toujours la même erreur.
j'ai même essayer sur une autre carte mère et j'ai toujours le même problème
j'ai changé d'ordinateur, câble USB aussi et toujours rien.
Si l'un de vous peut m'aider svp
Dans Arduino je sélectionne:

 Type de carte Sanguino
Processeur ATmega1284 or ATmega1284P 16Mhz
port COM3
Programmateur AVRISP mkII
Message d'erreur:
avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xae
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xae
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xae
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xae
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0xae
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xae
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xae
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xae
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xae
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xae

avrdude done.  Thank you.

3113Project, 2019-09-08:
have a bootloader in the board?

kinou33, 2019-09-09:
Bonjour
Oui j'ai un bootloader

3113Project, 2019-09-09:
avez-vous essayé de charger le programme en utilisant uniquement l'alimentation USB?
have you tried to load the program using only the usb power supply?

3113Project, 2019-09-09:
Try doing this: 1 card off unplugged from the pc, 2 open arduino, 3 load the program (arduino tells you that the card is not present), 4 connects the card, 5 selects the correct port, 6 tries to upload again.
Powering only from USB.

Essayez de faire ceci: 1 carte débranchée du PC, 2 arduino ouvert, 3 chargez le programme (arduino vous dit que la carte n’est pas présente), 4 connecte la carte, 5 sélectionne le bon port, 6 tente de télécharger à nouveau.
En se nourrissant uniquement par USB.

kinou33, 2019-09-09:
Hello,
First of all thank you for your reply.
Yes I powered the motherboard only with usb.
I will try the method of loading without the usb and I'll let you know.
sorry for my English ....

Bonjour,
tout d'abord, merci pour votre réponse.
Oui j'ai alimenté la carte mere uniquement avec de l'usb .
je vais essayer la méthode de charger sans l'usb et je vous tien au courant.

désolé pour mon anglais ....

kinou33, 2019-09-10:
Bonjour,
j'ai essayer la méthode que vous avez décrite mais c'est un echec.... ça c'est la mauvaise nouvelle.
La bonne nouvelle c'est que j'ai réussie a installer le firmware en passant par virtualbox ou j'ai une machine linux dessus.

Apres ça reste un grand mystere pourquoi windows ne veut plus communiquer avec ma carte .

Dans tout les cas, je vous remercie pour le temps que vous avez pris afin de m'aider

3113Project, 2019-09-10:
have you installed the drivers???

3113Project, 2019-09-10:
some board mount a ch34x comunication chip you can download on github or google it.

some are here: https://github.com/himalayanelixir/Arduino_USB_Drivers

kinou33, 2019-09-10:
Oui oui bien sur, j'avais deja modifier le firmware avant sous Windows ...c'est depuis la semaine dernière que l'erreur est apparu

Lauszus, 2019-09-21:
@kinou33 sounds like this issue was fixed, as it was a Windows driver issue?

@3113Project thanks for helping out!

kinou33, 2019-09-21:
Bonjour,
Oui le problème est résolu, merci à vous

closed 2019-09-21 by kinou33

Lauszus, 2019-09-21:
@kinou33 great to hear that :)