Lauszus/Sanguino #23: How can i fit an auto bed leveling sensor

GhassanYusuf, 2020-02-07:
i am trying to add an inductive sn04 sensor and want to use Marlin 2.0 i always get an error, can you please help me out

Ryder17z, 2020-02-08:
post the error so we can see what's happening

GhassanYusuf, 2020-02-08:
I am trying to fit marlin 2.0 On It

This is the picture of the board
[image]

This is the error
[image]

Ryder17z, 2020-02-10:
Does the file exist?

GhassanYusuf, 2020-02-12:
I don't know, but it compiles on the older version of marlin, why in marlin 2.0 it does not compile

Lauszus, 2020-02-12:
You should really post this question on one of the Marlin forums. This repository is simply so you can use your microcontroller with Arduino and has nothing to do with Marlin.

Lauszus, 2020-02-20:
I'm going to close this, as it has not really anything to do with this hardware add-on.

closed 2020-02-20 by Lauszus