Lauszus/Sanguino #24: Error while burning bootloader.

TechieAndroid, 2020-11-02:
I get this error:
```
avrdude: Expected signature for ATmega644 is 1E 96 09
        Double check chip, or use -F to override this check.
Error while burning bootloader. 
```
when I go to burn the bootloader on my Ender 3. 

I'm following a [guide](https://howchoo.com/ender3/ender-3-bootloader-firmware-update-marlin) to setup Marlin FW so that I can get thermal runaway protection and bltouch support.

The steps include using Sanguino as the board in the IDE with programmer set to Arduino as ISP (I already uploaded the ArduinoISP sketch to my Uno) and my Arduino as the flasher for my mainboard's bootloader.

[Screenshot](https://imgur.com/o3kBTuG)

TechieAndroid, 2020-11-02:
Closing because I figured it out and feel slightly stupid. There was a lack of information on the type of atmega the ender 3 has and once I changed it in the IDE it started working.

In my defense I'm new to arduino stuff still.

closed 2020-11-02 by TechieAndroid

Lauszus, 2020-11-02:
@TechieAndroid no problem. Good to hear that you got it working.