Lauszus/Sanguino #25: Unable to upload firmware to Creality V1.1.4 board using Sanguino ATMEGA1284p board in Arduino

Bobblebee, 2020-12-08:
Hello, 

I am trying to update the firmware of my Creality V1.1.4 board using Sanguino as an ATMEGA1284p board in Arduino. I've managed to burn a bootloader to the board, using an Arduino Uno, with programmer set as Arduino as ISP, and connecting the board to arduino via ICSP ports. The bootloader seems to upload fine (I already tried to reflash the bootloader multiple times, and all seems good in Arduino IDE).

However, when I try to upload the Firmware, it compiles fine, but it can't be uploaded to the board. These are the messages I get in arduino IDE:

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x01
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x01
... [all the way up to attempt 10]
avrdude done.  Thank you.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

I have tried multiple solutions.  So far, the best change was to modify the boards.txt located in C:\<USER FOLDER>\AppData\Local\Arduino15\packages\Sanguino\hardware\avr\1.0.3 with the following lines:

sanguino.upload.protocol=stk500
sanguino.menu.cpu.atmega1284p.upload.speed=115200

This got me as far as connecting to the board, but now I get timeout errors, and the following info on arduino IDE:
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer
         AVR Part                      : ATmega1284P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         65    10   256    0 yes    131072  256    512  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : STK500V2
         Description     : Atmel STK500
         Programmer Model: Unknown

I tried the following baudrates: 9600, 19200, 38400, 57600, 74880, 115200 and 250000. None of them worked. I also tried a different USB port (when the board is connected, the COM port appears in device manager, so I don't think it is a port or a cable issue). I honestly don't know what else to try. Can you help me?

Lauszus, 2021-02-07:
Sorry for the late reply.

Please verify that the bootloader is running by checking if the onboard LED (Pin PB0) turns on?

crossan007, 2021-02-07:
Any chance it's related to this: https://github.com/Lauszus/Sanguino/issues/26#issuecomment-766091817