Lauszus/Sanguino #26: Issue with optiboot_atmega1284p on Creality 1.1.4 board

crossan007, 2021-01-22:
I'm not 100% sure this is the right location to file the issue; but I'll start here.

I have a Creality Ender 3 Pro 1.1.4 board; which a few guides refer to as a Sanguino / ATMega1284p.

I followed the guide here https://www.fission3d.com/post/flash-bootloader-and-install-firmware-with-raspberry-pi to use the in circuit serial programmer (ICSP) to flash the Optiboot bootloader as obtained from the Arduino IDE Package manager at `C:\Users\USER\AppData\Local\Arduino15\packages\Sanguino\hardware\avr\1.0.3\bootloaders\optiboot` with avrdude.

The bootloader seems to have flashed fine, and worked for a period of time (a few weeks( using the klipper installation instructions further on in the guide linked above.

Recently, without explanation, I've been seeing a series of errors in the `dmesg` log of the Raspberry Pi into which this board is plugged:
```
[  449.874913] usb 1-1.3: new full-speed USB device number 3 using xhci_hcd
[  449.975205] usb 1-1.3: device descriptor read/64, error -32
[  450.195151] usb 1-1.3: device descriptor read/64, error -32
[  450.414919] usb 1-1.3: new full-speed USB device number 4 using xhci_hcd
[  450.515155] usb 1-1.3: device descriptor read/64, error -32
[  450.735213] usb 1-1.3: device descriptor read/64, error -32
[  450.855268] usb 1-1-port3: attempt power cycle
[  451.514915] usb 1-1.3: new full-speed USB device number 5 using xhci_hcd
[  451.515192] usb 1-1.3: Device not responding to setup address.
[  451.735116] usb 1-1.3: Device not responding to setup address.
[  451.954911] usb 1-1.3: device not accepting address 5, error -71
[  452.054921] usb 1-1.3: new full-speed USB device number 6 using xhci_hcd
[  452.055190] usb 1-1.3: Device not responding to setup address.
[  452.275150] usb 1-1.3: Device not responding to setup address.
[  452.494913] usb 1-1.3: device not accepting address 6, error -71
[  452.495436] usb 1-1-port3: unable to enumerate USB device
[  452.834924] usb 1-1.3: new full-speed USB device number 7 using xhci_hcd
[  452.935180] usb 1-1.3: device descriptor read/64, error -32
[  453.155152] usb 1-1.3: device descriptor read/64, error -32
[  453.374919] usb 1-1.3: new full-speed USB device number 8 using xhci_hcd
[  453.475161] usb 1-1.3: device descriptor read/64, error -32
[  453.695150] usb 1-1.3: device descriptor read/64, error -32
[  453.815434] usb 1-1-port3: attempt power cycle
[  454.474912] usb 1-1.3: new full-speed USB device number 9 using xhci_hcd
[  454.475188] usb 1-1.3: Device not responding to setup address.
[  454.695112] usb 1-1.3: Device not responding to setup address.
[  454.914920] usb 1-1.3: device not accepting address 9, error -71
[  455.014922] usb 1-1.3: new full-speed USB device number 10 using xhci_hcd
[  455.015188] usb 1-1.3: Device not responding to setup address.
[  455.235109] usb 1-1.3: Device not responding to setup address.
[  455.454919] usb 1-1.3: device not accepting address 10, error -71
[  455.455503] usb 1-1-port3: unable to enumerate USB device
```

If I plug the board into another Pi, or another computer, the board seems to function; so, yes - this may actually be a problem with my raspberry pi's USB ports.

What I can't explain is why _other_ USB devices continue to work on that pi:  only this device is failing to negotiate the USB connection.

Are there settings in the optiboot bootloader which define the USB host negotiation?

crossan007, 2021-01-23:
Somehow my board started working. 

All I did was get up to pour a glass of wine, and when I returned and plugged in the board, it negotiated a USB connection and I was able to flash Marlin and then Klipper

Ryder17z, 2021-01-23:
Sounds like a dodgy connection in the chain somewhere (cable?)

crossan007, 2021-01-23:
I actually think this came down to timing.

When I plugged in the USB within, say, 5 seconds of completing the avrdude ICSP flash, the USB negotiation failed (in the manner described above).

When I waited more like 15 seconds after completing the ICSP Flash to plug in USB, then the USB negotiation succeeded.

There still remains a minor issue, where, if the board is booted without a USB host connected, then plugging into a USB host at any time after boot will not yield a successful negotiation;  for instance, if I turn on my printer's power supply and allow the board to boot, the USB host will not recognize the board.  If I plug in the USB host before  turning on the printer, the power from USB is sufficient to boot up the board and establish a connection.

I don't know what component on the board is responsible for post-boot USB negotiation.  Any ideas?

Lauszus, 2021-01-24:
This issue is not related to this bootloader, as it is simply responsible for allowing the user to upload their firmware via UART and in this case does NOT handle any of the USB communication.

Other more advanced bootloaders does handle the USB communication, but in your case this is handled by the USB device to UART chip found on your board i.e. FTDI FT232R or similar.

Anyway please try to plug the board into another machine in order to verify that it's not an issue with your Raspberry Pi.