Lauszus/Sanguino #28: avr-g++: bad CPU type in executable on Mac M1

jvlobo, 2024-02-29:
I was trying to use the Sanguino (version 1.0.3) on the Arduino IDE (version 2.3.2) and I'm getting this error when it try to compile:

`Compilation error: fork/exec /Users/jvlobo/Library/Arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/avr-g++: bad CPU type in executable`

I was investigating about the `bad CPU type in executable` error and found this on the official Arduino page -> https://support.arduino.cc/hc/en-us/articles/7765785712156-Error-bad-CPU-type-in-executable-on-macOS

But even after installing Rosetta it does not work.

Any ideas of how to solve this?

Thanks a lot!

tomekskuta, 2024-07-28:
I found this fork with fix for m1 mac and it saved me 😮‍💨
https://github.com/wnagele/Sanguino

Knightmoved06, 2024-08-09:
New to this whole process, but I can't seem to get Arduino IDE to recognize the new fix file, would you be able to assist?

tomekskuta, 2024-08-09:
sure 👌 
if you already installed Sanguino you can remove it to make sure you don't use the one with no M1 support.
then:

- go to preferences (cmd+,) and in `Additional boards manager URLs` section paste `https://raw.githubusercontent.com/wnagele/Sanguino/master/package_lauszus_sanguino_index.json` - it's the json file from forked repo.
[image]
- go to your Boards Manager (cmd+shift+b) and find Sanguino. install it.
 
that should be all you need to do
[image]

Knightmoved06, 2024-08-15:
You are a God! Thank you! Running into other issues now, but I'll try to figure those out before I hassle you with those! Sorry for the delay in gratitude, just took me a hot minute to find some time to try this out. thank you!